technoblogy/ulisp #11: Cannot compile

JJ-Atkinson, 2018-09-24:
I'm trying to compile/upload ulisp, but I am running into an odd error. 

```
C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'object* myalloc()':

ulisp:187:44: error: 'error' was not declared in this scope

   if (Freespace == 0) error(PSTR("No room"));

                                            ^

C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'int saveimage(object*)':

ulisp:380:60: error: 'pfstring' was not declared in this scope

     pfstring(PSTR("Error: Image size too large: "), pserial);

                                                            ^

C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'int loadimage(object*)':

ulisp:400:74: error: 'error' was not declared in this scope

   if (imagesize == 0 || imagesize == 0xFFFF) error(PSTR("No saved image"));

                                                                          ^

C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'void error(const char*)':

ulisp:424:50: error: 'pfstring' was not declared in this scope

   pfl(pserial); pfstring(PSTR("Error: "), pserial);

                                                  ^

C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'void error2(object*, const char*)':

ulisp:431:50: error: 'pfstring' was not declared in this scope

   pfl(pserial); pfstring(PSTR("Error: "), pserial);

                                                  ^

C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'object* fn_break(object*, object*)':

ulisp:2315:39: error: 'pfstring' was not declared in this scope

   pfstring(PSTR("\rBreak!\r"), pserial);

                                       ^

C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'object* fn_gc(object*, object*)':

ulisp:2429:36: error: 'pfstring' was not declared in this scope

   pfstring(PSTR("Space: "), pserial);

                                    ^

C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'void supersub(object*, int, int, pfun_t)':

ulisp:2623:49: error: 'pfstring' was not declared in this scope

     if (atom(form)) { pfstring(PSTR(" . "), pfun); printobject(form, pfun); pfun(')'); return; }

                                                 ^

C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'object* eval(object*, object*)':

ulisp:3171:72: error: 'pfstring' was not declared in this scope

       printobject(fname, pserial); pfstring(PSTR(" returned "), pserial);

                                                                        ^

C:\Users\Jarrett\Documents\Arduino\ulisp\ulisp.ino: In function 'void pcharacter(char, pfun_t)':

ulisp:3214:23: error: 'pfstring' was not declared in this scope

       pfstring(p, pfun);

                       ^

exit status 1
'error' was not declared in this scope
```

I copied the code directly from the `ulisp.ino` file into a new project. Moving the definition of `error` above the statements in question removes that error. I have tried compiling on the latest Arduino version on both Windows and Ubuntu.

technoblogy, 2018-10-04:
Sorry for the delay in responding.

This is a problem that seems to have appeared with the latest version of the Arduino IDE. The fix is to locate the comment:

    // Forward references

and insert the following two lines after it:

    void error (PGM_P string);
    void pfstring (PGM_P s, pfun_t pfun);

This fix will be incorporated in the next release.

Explanation: normally the Arduino IDE automatically resolves forward references to functions that are defined later in the file, but for some reason it has stopped working for these two references.

JJ-Atkinson, 2018-10-08:
Interesting. Is the error thrown by an Arduino Pre-Processor, and not the avr compiler? Thanks!

closed 2018-10-08 by JJ-Atkinson

technoblogy, 2018-10-09:
I think the compiler. It's fixed in Version 2.4.