technoblogy/ulisp #24: Arduino UNO bugged?

aaronjeline, 2019-05-27:
I tried uploading the ulisp source to my arudino uno. 
First, I uploaded a simple blinking program to test, arduino worked fine.
The upload got stuck in "Programmer not responding" 
Now, both the On and L LED are always lit, reset does nothing, and a cannot upload anything?
Any advice?

technoblogy, 2019-05-27:
What system are you running the Arduino IDE on (Mac, PC, Linux)?

Is the USB port selected in the Tools->Port menu?

aaronjeline, 2019-05-27:
Linux, yes. Uploading currently gives me the "out of sync" message.

technoblogy, 2019-05-27:
I don't have experience of using the Arduino IDE on Linux, but this error is to do with the USB communication. Have you tried relaunching the Arduino IDE?

On one forum I've read that it can help to unplug the USB, press the Upload tool, and just when the "Binary sketch size: xxx bytes" message appears reconnect the USB plug.

Let me know how you get on.

aaronjeline, 2019-05-27:
Trying that now, other people suggested reflashing the bootloader since reset does nothing. Thanks for your help and your project looks super cool

technoblogy, 2019-05-27:
Unless you chose the "Burn Bootloader" menu option by mistake I think it's very unlikely the bootloader has been overwritten. It's normally protected from being overwritten by an uploaded program. This sounds more like a simple USB communication problem.

aaronjeline, 2019-05-27:
Your suggestion did not work, same response. Nothing done gets the arduino to respond in any way to any signal I give it.

technoblogy, 2019-05-27:
How many USB ports have you got?

aaronjeline, 2019-05-27:
Both have been tried.

technoblogy, 2019-05-27:
Have you used the Arduino with your computer before (apart from the Blink upload)? Have you got another computer you can try it on?

aaronjeline, 2019-05-27:
Yes I have, it's always worked fine, no I do not. Thanks for you help, but it's looking pretty dead.

technoblogy, 2019-05-31:
Any progress with sorting this out?

aaronjeline, 2019-06-03:
Yeah.
Uploaded to a different arduino, 0 problems.
Used that arduino the reflash bootloader in last attempt to rescue, nothing
Freak accident I guess. Super weird. Thanks for all your advice!

closed 2019-06-03 by aaronjeline