technoblogy/ulisp #48: Problem compiling 4.0a for ATmega1284p using MightyCore 2.1.3 in Arduino 1.8.15

markhellewell, 2021-08-31:
Hello! I'm just starting to tinker with this great looking project and have hit a bit of a snag. Compilation fails for me with the following error

`In file included from C:\Users\me\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3\cores\MCUdude_corefiles/Arduino.h:28:0,
                 from sketch\ulisp.ino.cpp:1:
C:\Users\me\Documents\Arduino\ulisp\ulisp.ino: In function 'void supersub(object*, int, int, pfun_t)':
ulisp:1703:24: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'builtin_t' [-fpermissive]
       if (sname == sym(pgm_read_byte(&ppspecial[i]))) { special = 1; break; }
                        ^
C:\Users\me\Documents\Arduino\ulisp\ulisp.ino:854:10: note:   initializing argument 1 of 'symbol_t sym(builtin_t)'
 symbol_t sym (builtin_t x) {
          ^~~
In file included from C:\Users\me\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3\cores\MCUdude_corefiles/Arduino.h:28:0,
                 from sketch\ulisp.ino.cpp:1:
C:\Users\me\Documents\Arduino\ulisp\ulisp.ino: In function 'void pbuiltin(builtin_t, pfun_t)':
ulisp:4343:13: error: invalid conversion from 'uint16_t {aka unsigned int}' to 'const char*' [-fpermissive]
   PGM_P s = pgm_read_word(&lookup_table[name].string);
             ^
exit status 1
invalid conversion from 'uint8_t {aka unsigned char}' to 'builtin_t' [-fpermissive]`

technoblogy, 2021-08-31:
Sorry, I had a new version that should fix those issues, but hadn't got around to uploading it. It's there now - let me know if it helps. Thanks, David

closed 2021-08-31 by technoblogy

markhellewell, 2021-08-31:
@technoblogy wonderful, that fixed it! thanks so much :)

commit 1d18eafe (2021-08-31): Version 4.0b - 31st August 2021

Fixes #48