technoblogy/ulisp #50: I can't use with-output-to-string function

roscoe-tw, 2022-04-13:
I can't use with-output-to-string
I installed on Arduino UNO and Arduino MEGA using uLisp 4.1a

>(WITH-OUTPUT-TO-STRING (STR) (FORMAT STR "~A" "test"))
Error: undefined: WITH-OUTPUT-TO-STRIN

Is this function name too long? In addition, when using (save-image), if the function name is more than three characters, after resetting and using (load-image), the program will not be defined.

technoblogy, 2022-04-13:
Thanks for reporting this. The function **with-output-to-string** isn't defined in the AVR version of uLisp due to space limitations. This should have been documented in the Reference; sorry about that - I'll make sure it's added.

As a workaround you may be able to achieve what you want by outputting to a string from **format**.

> In addition, when using (save-image), if the function name is more than three characters, after resetting and using (load-image), the program will not be defined.

I can't reproduce this. Can you give an example?

technoblogy, 2023-04-14:
As of uLisp Release 4.4 with-output-to-string is included in all versions of uLisp, including the AVR versions.

closed 2023-04-14 by technoblogy