technoblogy/ulisp #56: ulisp-nano.ino is too big for ATmega328P with bootloader

dragoncoder047, 2023-05-19:
```
Sketch uses 32644 bytes (101%) of program storage space. Maximum is 32256 bytes.
Global variables use 1536 bytes (75%) of dynamic memory, leaving 512 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Arduino Uno.
```

I had to manually delete some code (I got rid of the saveimage/loadimage functions) to be able to run the tests in https://github.com/technoblogy/ulisp-esp/issues/75#issuecomment-1555183068.

technoblogy, 2023-05-19:
I'm puzzled because I tested it on an Arduino Uno. Which version of uLisp is it? What platform are you on - Arduino IDE version and OS?

dragoncoder047, 2023-05-19:
Arduino 1.8.13 on a raspberry pi, and it's literally a stock Arduino Uno R3. I just copypasted the entire ulisp-nano.ino from the repository into a new sketch.

If I was able to program it using ISP (sans bootloader), that would give me an extra 512 bytes, and it would fit. Shucks.

technoblogy, 2023-05-19:
When I upload uLisp AVR-Nano Release 4.4b on Arduino IDE 1.8.19 on my Mac, via the bootloader, I get:
````text
Sketch uses 32244 bytes (99%) of program storage space. Maximum is 32256 bytes.
Global variables use 1536 bytes (75%) of dynamic memory, leaving 512 bytes for local variables. Maximum is 2048 bytes.
````
So I suppose it must be a platform difference. You'll just have to comment out some function you don't need.

dragoncoder047, 2023-05-19:
That's funny. The Arduino IDE says my "Arduino AVR Boards" package is up-to-date.

closed 2023-05-19 by dragoncoder047

technoblogy, 2023-05-20:
The difference between yours and mine is 400 bytes. I wonder what could be causing that. Perhaps different compiler optimisation setting?

dragoncoder047, 2023-05-20:
My guess is between 1.8.13 and 1.8.19, avr-gcc had an update.

For boards where fitting in flash is a concern, I think `#pragma GCC optimize ("Os")` might help since you can't specify compiler flags in Arduino.