technoblogy/ulisp #58: Can't escape from delay

technoblogy, 2023-08-16:
If you evaluate a delay, such as:

    (delay 10000)

you can't escape.

dragoncoder047, 2023-09-20:
The easiest solution I can think of is to replace the single `delay()` call with a while loop

```cpp
unsigned long ms = /* time to delay */;
while (--ms) {
  delay(1);
  testescape();
}
```

technoblogy, 2023-09-21:
Good suggestion. More accurate is:
````text
object *fn_delay (object *args, object *env) {
  (void) env;
  object *arg1 = first(args);
  unsigned long start = millis();
  unsigned long total = checkinteger(arg1);
  do testescape();
  while (millis() - start < total);
  return arg1;
}
````

closed 2023-09-21 by technoblogy

dragoncoder047, 2023-09-21:
Ah, I forgot that testescape() takes a small, but significant, amount of time to run.