technoblogy/ulisp #59: `push` doesn't need to be a macro (nit pick)

dragoncoder047, 2023-09-20:
Not sure if one of the goals of uLisp is to be easily understandable, but since Arduino is C++ why not take advantage of it?

```cpp
//                               vv----------& makes the parameter a pass-by-reference
void push (object *thing, object *&stack) {
  object *cell = cons(thing, stack);
  stack = cell; // because stack is pass-by-reference this updates the caller's variable
}
```

technoblogy, 2023-09-21:
Would that be more efficient? The advantage of the current way using #define is that the C versions of all the Lisp primitives, such as push, pop, car, cdr etc, are in the same place.

dragoncoder047, 2023-09-21:
Well, (a) I doubt it would make any difference especially if you declare it as `inline void` (macros are inlined by definition). `twist()` and `untwist()` can also be implemented as functions instead of macros. I frankly like the function (not macro) form because you can see the types of the variables, can give them more descriptive names without having to worry about a name conflict, and you don't need extra parentheses around every macro parameter.

dragoncoder047, 2023-09-21:
> I'm (slowly) working on a port of ulisp to RISC-V (HiFive Inventor),

David already has a RISC-V port of uLisp, if you want to use that as a starting point: https://github.com/technoblogy/ulisp-riscv