technoblogy/ulisp #65: `(apropos)` doesn't show the right type for tail forms

dragoncoder047, 2024-06-14:
```
> (apropos 'progn)
progn (symbol/keyword)
```

Fix in `apropos()`:

```diff
   // Built-in?
   int entries = tablesize(0) + tablesize(1);
   for (int i = 0; i < entries; i++) {
     if (findsubstring(part, (builtin_t)i)) {
       if (print) {
         uint8_t fntype = getminmax(i)>>6;
         pbuiltin((builtin_t)i, pserial); pserial(' '); pserial('(');
         if (fntype == FUNCTIONS) pfstring(PSTR("function"), pserial);
         if (fntype == FUNCTIONS) pfstring(PSTR("function"), pserial);
-        else if (fntype == SPECIAL_FORMS) pfstring(PSTR("special form"), pserial);
+        else if (fntype == SPECIAL_FORMS || fntype == TAIL_FORMS) pfstring(PSTR("special form"), pserial);
         else pfstring(PSTR("symbol/keyword"), pserial);
         pserial(')'); pln(pserial);
       } else {
         cdr(ptr) = cons(bsymbol(i), NULL); ptr = cdr(ptr);
       }
     }
     testescape();
   }
   return cdr(result);
 }
```

technoblogy, 2024-06-14:
I see your point, but "TAIL FORMS" is not really an official Lisp term; I use it internally to distinguish special forms that support tail recursion, but it's not used in the user documentation; for example:

http://www.ulisp.com/show?3L#progn

so for that reason I think **apropos** shouldn't make a distinction.

dragoncoder047, 2024-06-14:
I wasn't asking for it to print "tail form" for progn, and either way, the fix prints "special form" for both tail forms and special forms since that's how the evaluator treats them (aside form the tail call functionality).