technoblogy/ulisp #73: Feature requist: add link to installation instructions in webpage sidebar menu

E3D3, 2024-08-16:
Hi,
this is more a suggestion than a feature request.

After reading about this project on "http://www.ulisp.com" to find out if it is something for me, I looked for the installation instructions; first in the menu section "Getting started", then I looked for it on the startpage and the tutorial [Getting started](http://www.ulisp.com/show?1AAO), and finally found them on page "Download uLisp". Is see similar placement often with other software but still think that it would be helpfull for many people to really get started, to also add a link to installation instructions to the sidebar menu.

Best regards.

technoblogy, 2024-08-16:
Thanks for your suggestion. So do you mean another page called **Installation instructions** under the **Getting started** heading in the sidebar menu, perhaps just above **Performance**?

E3D3, 2024-08-16:
I think it is enough to add only a link, like you suggested, optionally add it also to the sentence on the [start page](http://www.ulisp.com/)

> You can download the current version of uLisp free from the (Download uLisp)[http://www.ulisp.com/show?1AA0] page.

so it becomes even more helpful, for example like (forgive my bad English):

> You can find downloads and installation instructions for the current version of uLisp free on the [Download uLisp](http://www.ulisp.com/show?1AA0) page.

closed 2024-08-16 by technoblogy

technoblogy, 2024-08-16:
Please check what I've done and see if it's an improvement.

Thanks! David

E3D3, 2024-08-25:
sorry for my late response.

IMO it is a very helpful improvement.
Thank you for the friendly response & super quick action.